Xinyuan-LilyGO/T-Internet-COM #2: Ethernet not working

jfnunez-dev, 2022-03-03:
Hi, i am using the lilygo T-Internet-COM V1.1, i am trying to run the ethernet demo (ETHDemo.ino) but I am getting this and the ethernet isn't working:
 
E (2036) emac: GPIO16 and GPIO17 has been occupied by PSRAM, Only ETH_CLOCK_GPIO_IN is supported!

which its the solution to this problem?
Salutes

LilyGO, 2022-03-17:
May I ask if you are using a board to send pictures for me to observe?

LilyGO, 2022-08-02:
The problem is too long, you can open it again if necessary

closed 2022-08-02 by LilyGO

clederson-Ms, 2022-08-08:
Boa noite!!
Estou com o mesmo problema, estou no Brasil existe alguma configuração deste codido ETH_CLOCK_GPIO_IN??
Percebi que a hora no display está errada...
Obrigado

LilyGO, 2022-08-08:
请确保你的ARDUINO的ESP32框架为2.0.3版本

LilyGO, 2022-08-08:
Please make sure your ARDUINO's ESP32 framework is version 2.0.3

clederson-Ms, 2022-08-08:
Não é a versão 1.0.6
[image]

clederson-Ms, 2022-08-12:
Atualizei a estrutura do ESP32 para a versão 2.0.3, porém estou com um novo problema, poderia ajudar por favor??

E (2837) esp.emac: emac_esp32_init(349): reset timeout
E (2838) esp_eth: esp_eth_driver_install(214): init mac failed

LilyGO, 2022-08-13:
Please configure it like this.

> #define ETH_CLK_MODE ETH_CLOCK_GPIO0_OUT