Xinyuan-LilyGO/T-Internet-COM #5: WiFi-AP function forwarding over Lte

etempm, 2022-03-21:
Hi, I can not find example about WiFi-AP function forwarding over 4g lte modem (sim7600a). This is urgent for a project. Can you share example about this?

LilyGO, 2022-03-22:
https://github.com/Xinyuan-LilyGO/T-Internet-COM/tree/main/example/ESP-IDF/pppoe2ap

sanjulanipun, 2022-04-02:
> https://github.com/Xinyuan-LilyGO/T-Internet-COM/tree/main/example/ESP-IDF/pppoe2ap

Hi im working with this board, i need to know can we run it arduino ide, because i need to run that code with my own some funtions. It have arduino custom made special library it only work with arduino ide. Are you possible help me for that.

I just only need your funtion get into arduino ide.

LilyGO, 2022-05-27:
ESP32 also uses IDF in the bottom layer of Arduino. You can import entire component projects into your library.

LilyGO, 2022-08-02:
The problem is too long, you can open it again if necessary

closed 2022-08-02 by LilyGO