Xinyuan-LilyGO/T-Internet-COM #8: A few questions

gmankab, 2022-05-28:
I am in Russia, and I am going to put a SIM card from the Russian operator MTS into this device.

Does this device have IMEI and will it work in Russia? I'm going to leave Russia for Europe because of the war with Ukraine, so will this device work in Europe with European operators?

If I use a USB-C to USB-A cable to connect this device to a computer, will the Internet be transferred to the computer without any problems? Will I need to do any additional settings? If I connect this device to the router via ethernet, will the Internet be transferred to the router?

Does this device support 4G? What is the maximum internet speed this device can transfer over USB-A, USB-C, and ethernet?

Is it possible to connect an external ssd with arch linux to this device and use it as a server for downloading files from the Internet?

LilyGO, 2022-05-31:
This product does not meet all the needs you describe. Its IMEI, IMSI depends on the PCIE module and SIM card it is paired with. PCIE equipment can be used when the area is supported. USB-C to USB-A is only used for serial uploading programs and DEBUG debugging, and cannot access the Internet through this cable.

LilyGO, 2022-08-02:
The problem is too long, you can open it again if necessary

closed 2022-08-02 by LilyGO