Xinyuan-LilyGO/T-Internet-COM #10: Programing Procedure for T-Internet-COM

unklstewy, 2022-08-01:
When attempting to program the onboard ESP-WROVER-32 with platformio, I am unable to do so successfully.  esptool keeps reporting an error:
Retrieving maximum program size .pio/build/esp-wrover-kit/firmware.elf
Checking size .pio/build/esp-wrover-kit/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.0% (used 39188 bytes from 327680 bytes)
Flash: [====      ]  35.9% (used 752518 bytes from 2097152 bytes)
Building .pio/build/esp-wrover-kit/firmware.bin
esptool.py v3.1
Merged 1 ELF section
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, ftdi, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/cu.usbmodem54350137791
Uploading .pio/build/esp-wrover-kit/firmware.bin
esptool.py v3.1
Serial port /dev/cu.usbmodem54350137791
Connecting.......
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: e8:31:cd:38:7c:84
**Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000)
*** [upload] Error 2**

I have followed previous guidance suggesting that maybe the USB cable was bad and switched out multiple cables, or that the TTL USB transceiver was not keeping up with the data rate and attempted to slow the baud rate down gradually to 9600 from 115200.  Any speed other than 115200 would fail to connect to the port.  

I have tried tying GPIO0 (pin 25) to ground and then pressing reset to put the mcu into 0x02 boot mode and have been able to get the "ready for download" message, but I can not get it to move past the fatal error ' Failed to write to target ram'.

Build Environment:
Microsoft Visual Code : Version: 1.69.2 (Universal)
PlatformIO: Version Core 6.1.3 Home 3.4.3
OperatingSystem: MacOS Monterey v12.2.1

unklstewy, 2022-08-01:
To be clear, I have tried both  board=esp32dev  and board = esp32-wrover-kit each in their own environments.

LilyGO, 2022-08-02:
Please use flash_download_tool to erase the chip of the module.