Xinyuan-LilyGO/T-Internet-COM #13: Esp32 in looping

Tonymichelon, 2023-01-30:
Hello, I would like to know which name of the board in the Arduino IDE should I select?
After I put the read.me link, my IDE has a huge list of esp32 boards, I selected wt32-eth01 but my esp is looping... I use Arduino IDE 2.0.3

mmMicky, 2023-01-31:
Select ESP32 DEV MODULE

Tonymichelon, 2023-01-31:
I selected the ESP32 DEV MODULE, but I keep having looping problem, below what is coming out in the serial monitor repeatedly:

rst:0x8 (TG1WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
ets Jul 29 2019 12:21:46

mmMicky, 2023-02-01:
It's possible that the code is causing the reset. You try to compile another simple example for testing.

Tonymichelon, 2023-02-10:
i fixed a problem in the code, but now i have another error:

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
E (51) lan87xx: lan87xx_init(499): wrong chip ID
E (52) esp_eth: esp_eth_driver_install(215): init phy failed