Xinyuan-LilyGO/T-Internet-COM #16: what is the AUX antenna connector for?

beta-tester, 2023-02-13:
hi,

on a PCIE-SIM7600x module there are GNSS, MAIN and AUX connectors.
GNSS and MAIN i know what it is good for, but what is AUX?

mmMicky, 2023-02-15:
auxiliary antenna. Used to assist LTE or 2G signal acquisition.

beta-tester, 2023-02-15:
you mean as secondary LTE antenna?

i found a manual of all AT commands of the SIM7600 module and it looks like it has its own WiFi & BT built in.
i thought AUX is maybe an antenna connector for that Wifi & BT of  the SIM7600 module, beside the WiFi & BT of the ESP32 module.

mmMicky, 2023-02-16:
for example. Regardless of sim800L or other signals, it will have an AUX interface. And SIM800L does not have WIFI function.

beta-tester, 2023-02-16:
that's strange.
why does SIMCOM list AT commands for WIFI and BT in their SIM7600 manual (SIM7500_SIM7600 Series_AT Command Manual_V3.00.pdf),
but the SIM7600 module itself does not have WIFI/BT functioallity (SIM7600 Series_SPEC_202111.pdf).
or does that SIM7600 module support WIFI/BT over its USB-OTG?

on the T-Internet-COM module there are the V_BUS, D+, D-, GND lines of the SIM7600 module as test pads populated.
in the schematics it looka V_BUS on the PCIE-Module goes to V_MAIN (+4V) and on the T_Internet-COM module V_BUS goes to the USB-C connector (+5V).

i will not destroy anything...
how can i connect the V_BUS, D+, D-, GND test pads correctly to hook up PCIE-SIM7600 to a PC to DEBUG the module, upload SIM7600-Firmware, use its RNDIS functionallity?

sorry when my questions are stupid - i try to understand the PCIE-SIM7600 module and its capabilities

beta-tester, 2023-02-16:
on the T-Internet-COM module there are the V_BUS, D+, D-, GND lines of the SIM7600 module populated as test pads.
in the schematics it looka V_BUS on the PCIE-Module goes to V_MAIN (+4V) and on the T_Internet-COM module V_BUS goes to the USB-C connector (+5V).

i will not destroy anything...
how can i connect the V_BUS, D+, D-, GND test pads correctly to hook up PCIE-SIM7600 to a PC to DEBUG the module, upload SIM7600-Firmware, use its RNDIS functionallity?

sorry when my questions are stupid - i try to understand the PCIE-SIM7600 module and its capabilities

closed 2023-02-16 by beta-tester